josephernest/SamplerBox #2: Timed sustain

maditnerd, 2015-06-17:
(This is mostly a reminder for me)

I needed some sustain for the piano but if I use sustain=True there was too much sustain.
You can however have more or less sustain by tweaking FADEOUTLENGTH.

josephernest, 2015-06-18:
In fact `sustain = True` is modified "live" when we use a Sustain Pedal.
When a Midi Message CC #64 arrives (see [here](http://nickfever.com/music/midi-cc-list)), sustain is modified to True or False :
https://github.com/josephernest/SamplerBox/blob/master/samplerbox.py#L218 
If we manually set `sustain = True`, all the notes will last forever, and this results in too much sustain.

The best way is probably, as you said to tweak `FADEOUTLENGTH`... 

... or maybe later we should add more things in `definition.txt`,  such as %fadeout  or %release  etc.

SirPrimalform, 2017-04-15:
A per-instrument fade out time in the definition file would be much appreciated. The current global setting means it'll always be wrong for some instruments. Organs and mellotron sounds need almost immediate fade out, but that obviously sounds weird for a natural sounding piano.