josephernest/SamplerBox #6: Nothing Happening

brightmoon03, 2015-07-02:
When I put the .IMG file onto my sd card, then put it in my RPI 2, I plugged it in, but nothing happened. My keyboard didn't turn on when plugged in through USB, but my speaker had a faint static-like sound.
I am not using the full setup, just with my midi keyboard over USB.
Can you help with this?

Thanks, Samuel

Hiradur, 2015-07-30:
Perhaps the Pi does not provide enough power for the keyboard?

josephernest, 2015-07-31:
Hi @brightmoon03 , are you sure the .IMG file has been properly written to the microSD card? With which tool did you do it? 
Have you tried to either connect a monitor to see what is displayed, or to connect to the Pi via SSH ?

pizaranha, 2020-03-27:
Hi @samuelpianoguy, Today I had the same problem with the iRig Keys USB MIDI Keyboard. I try another USB cable and it worked!

Thanks @josephernest for such versatil tool.

josephernest, 2022-08-08:
Thanks @pizaranha! @samuelpianoguy I'm sure you found the solution in the meantime, so I close this issue :)

closed 2022-08-08 by josephernest