josephernest/SamplerBox #19: Feature Request: Randomise sample

mattgreenfield, 2016-06-06:
Hi,

I'm using sampler box to trigger drum sounds from my electronic drum kit. The sample sets I am using (acoustic drum kits) have 3 hits for each drum / cymbal, each hit is the same velocity but varies in tone slightly and helps make the kit sound more natural and less like a drum machine.

Could there be a way to have 3 .wav files for the same note that randomise?

Thanks for the great software by the way.

maditnerd, 2016-09-05:
:+1: 
It would be interesting to be able to define this behaviour in definition.txt

Here is some thought on how this could be made, i'll try it when I can.

I think that would require to modify the variable : ```samples```
As for now samples has two argument ```midinote``` and ```velocity```
We would need to add a ```third argument``` for example ```randomize```

dhilowitz, 2016-11-08:
For notes with 4 or more samples, I would especially like it to be randomize without consecutive repeats.  I currently use a custom script to do this in Kontakt. 

Examples:

True random might look like this: 1-4-3-3-2-4-1-1-3-4.
Modified random would look like this: 1-4-3-2-1-3-2-4.

dhilowitz, 2016-11-14:
I've now made my own fork that contains sample randomization: https://github.com/dhilowitz/SamplerBox. 

@josephernest: I'd be happy to do a pull request, if you're interested.

josephernest, 2016-11-30:
@dhilowitz Hi David, Thanks for this addition. We are in a process of merging the additions. I'll send you an email very soon.