josephernest/SamplerBox #20: Sample never ends or lasts too much time

stef35, 2016-06-29:
Sometimes, when i play default 0 saw files, it never ends. I need to restart my roland module.
It may happen as well that the sample lasts more than expected.
And some other times, it works fine.

my config: rpi3, jessie dist with manual installation of the paul's implmentation (i need to run another appli on it), usb/midi convertissor, roland td6 module

Any idea?
Cheers
ps: i have noticed a lot of midi messages sent by the module (maybe clock messages) regularly

stef35, 2016-06-30:
some update,
i have noticed by commenting the block "note off" the script, we always got the problem (0 saw files never end). That means that sometimes, for any reason (maybe the lot of 530/531 midi messages received), the "note off" is not catched by the programm.

josephernest, 2022-08-08:
@stef35 Out of curiosity, did you find a solution in the meantime? Was it something specific to your Roland module?