josephernest/SamplerBox #24: Serial midi on recent images

Cuki79, 2017-01-20:
Cuki posted Jan 18, 22:02:

I confirm what was also observed by r.meurer (posted Nov 17, 16:53:) and jpg (posted Jan 8, 23:04:) on the samplerbox forum

I could make serial MIDI work with samplerbox_20150618.img on RPI 2B

I could NOT make serial MIDI work with samplerbox_20161228.img on RPI 2B or RPI 3

Using USB-Midi interface and Akai LPK25 works with samplerbox_20161228.img on RPI 2B and RPI 3.

note: RPI 3 problem is probably due to the fact that bluetooth takes the full serial port that was available in RPI2. For RPI2 problem I have no explanation.

josephernest, 2017-01-31:
Thanks for this report. 

Until this is solved : previous image : http://samplerbox.org/files/images/samplerbox_20150618.zip

gkita, 2017-02-09:
Some searches on google said, the bluetooth is using the uart, so need to disabled bluetooth.

In the /boot/config.txt:
dtoverlay=pi3-disable-bt

reboot, then uart start to work, but only 38400 bit/s, not 31250. (I build an easy tester with arduino, and on RPI run an easy python script, which is print the received bytes)

It is not enough:

/boot/cmdlin.txt:
bcm2708.uart_clock=3000000

/boot/config.txt: :
init_uart_clock=2441406
init_uart_baud=38400

The uart is still working only 38400 baud.

The solution:
Need an other line in the /boot/config.txt:
dtparam=uart0_clkrate=3000000

Reboot, and wow, uart is working 31250 baud, and the samplerbox have sound, when I play on the midi keyboard :) (Not with usb-midi, but raspberry's physical uart with optocoupler, like this: http://samplerbox.org/makeitelectronic)

-------------------------------------------------------------------------------------------------------
In summary, my /boot/cmdline.txt:

root=/dev/mmcblk0p2 ro rootwait console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 elevator=noop bcm2708.uart_clock=3000000

and my /boot/config.txt:

device_tree_param=i2c_arm=on
init_uart_clock=2441406
init_uart_baud=38400
gpu_mem=64
boot_delay=0
disable_splash=1
disable_audio_dither=1
dtparam=audio=on
dtoverlay=pi3-disable-bt
dtparam=uart0_clkrate=3000000

Cuki79, 2017-02-10:
Thanks it works. I get some bug when I push the key to fast. But for my purpose it works.

gkita, 2017-02-10:
Hi,

I'm glad :)

I use 6N138 optcoupler, and it make
no mistake.
I think, the optocoupler to be slow. (4N35 etc too slow)
Or (with 6N138) try 470 ohm collector resistor (not 1k), and 4.7 kOhm
instead of 10k. The optocoupler will faster off.
�
Other
thing. Do you have any idea, how can I decrease audio latency?� I
think (Pi3), it is 20-30 ms (according hearing). It would be good 4-5
ms.

Thank you!

Best regards.
Gabor 

> Thanks it works. I get some bug when I push the key to
fast. But for my
> purpose it works.
> 
> --
> You are receiving this because you commented.
> Reply to
this email directly or view it on GitHub:
>
https://github.com/josephernest/SamplerBox/issues/24#issuecomment-279075497

Cuki79, 2017-02-11:
I've used a 6N137 with different resistors... May be that is the reason why.

Sorry but I have no idea for the latency.

I bought my first RPI at christmas and did not have much time to mess with it.

Regards,
Kien

josephernest, 2017-05-16:
I confirm this solves the problem for old Linux versions (<= 4.4). For newer versions, all we have to do is to add this to `/boot/config.txt`:

    enable_uart=1
    dtoverlay=pi3-miniuart-bt
    dtoverlay=midi-uart0 

(and remove `init_uart_clock=2441406`, `init_uart_baud=38400` in `config.txt` and `bcm2708.uart_clock=3000000` from `cmdline.txt`)

See http://www.samplerbox.org/article/midiinwithrpi.

closed 2017-05-16 by josephernest