josephernest/SamplerBox #30: Not Working - "ImportError: No module named sounddevice"

bkleeman, 2017-12-03:
Hey y'all,

I'm having some trouble getting SamplerBox to work on my Raspberry Pi 3, running Raspbian. I am using an Arturia MiniLab Mk II, connected to the Pi via USB. I did not build the circuit or the enclosure - I am just trying to use the software for the time being.

I followed the instructions from the [README](https://github.com/josephernest/SamplerBox) under the Install section by pasting each line into the terminal on my Pi. 

After the line `sudo pip install rtmidi-python pyaudio cffi sounddevice`, a couple .tar.gz files were successfully downloaded, but then it said "collecting cffi" and then in red listed lines and lines of exceptions. At the end of that output, it says, again in red `TypeError: unsupported operand type(s) for -=: 'Retry' and 'int`

The next command seemed to go successfully. 

After entering `python samplerbox.py`, I receive this output:
`Traceback (most recent call last):
  File "samplerbox.py", line 35, in <module>
    import sounddevice
ImportError: No module named sounddevice
`
I'm not sure if I followed the instructions incorrectly, or what the problem might be. I'm a total newbie to Raspberry Pi and the Linux ecosystem, so I wouldn't be surprised if I made a mistake that would be obvious to folks with more experience. Any help is greatly appreciated!

Thanks for making awesome software!

dkadrios, 2019-01-31:
Any update on this?  I'm running into exactly the same issue on Pi 3 and Raspbian

josephernest, 2019-01-31:
You probably need to install https://pypi.org/project/sounddevice/ with:
pip install sounddevice

dkadrios, 2019-02-05:
I hadn't tried installing the prerequisites individually, just all at once per the readme `sudo pip install rtmidi-python pyaudio cffi sounddevice`

I started fresh yesterday with the latest raspbian and it's working fine now (of course)  🙄

josephernest, 2019-02-05:
Cool :)

closed 2019-02-05 by josephernest

baranskistad, 2019-07-20:
Can confirm, running `sudo pip install rtmidi-python pyaudio cffi sounddevice` fixes problem.