josephernest/SamplerBox #37: New pi (2019) fails to boot

lazyoldjack, 2019-06-10:
Hi Joseph,

SamplerBox is a wonderful creation. Thank you for making and sharing!

I think that samplerbox_20170710 fails to boot in my 2019 Rpi because (I'm guessing) it's built on an OS that is now incompatible with the latest Raspberry Pi (which has some hardware change).

Please can you rebuild the load-and-go download using the latest OS? I'm sure that there will be many people who will be very grateful and it's a shame for this excellent project to not have a much longer life.

Thanks!

John

SoundGuy, 2022-04-24:
i have also tried to run the smaplerbox image from the website on my rpi and it didn't work. 
i managed to get it set up with python using this git on a raspbian.

josephernest, 2022-08-08:
@SoundGuy @lazyoldjack This is solved with the new image which is available on the website, you probably found it in the meantime. Alternatively, you can use a default RaspiOS image, and install SamplerBox manually [with this method](https://github.com/josephernest/SamplerBox#install).

closed 2022-08-08 by josephernest