josephernest/SamplerBox #39: ModuleNotFoundError: No module named 'sounddevice'

sharonosahon, 2019-11-14:
Hi,

Trying this line of code on jupyter notebook:

`import sounddevice as sd`

Getting the following error:

`ModuleNotFoundError: No module named 'sounddevice'`

Have tried:

`pip install sounddevice`
`conda install -c conda-forge python-sounddevice`
`brew install portaudio`
`brew reinstall portaudio`
`pip install setuptools --user`
`pip install cffi --user`
`sudo pip install rtmidi-python pyaudio cffi sounddevice`
`pip install --upgrade pip`
`python3 -m pip install sounddevice`

Does anyone know what might be wrong?
Thank you in advance!

sharonosahon, 2019-11-14:
Have also just tried 
`conda install -c conda-forge/label/gcc7 python-sounddevice`
`conda install -c conda-forge/label/cf201901 python-sounddevice`

but no luck

josephernest, 2022-08-08:
@sharonosahon SamplerBox is not really made to be used inside a Jupyter Notebook (well I have never tested, but it would be strange!). Rather it should be started in script mode as a forever running process, for example in command line:

    python samplerbox.py

Also custom config with Anaconda / conda is not really supported - everything is designed to run with the standard `python`.

closed 2022-08-08 by josephernest