josephernest/SamplerBox #49: MIDI 38400 / 31250 baud hack

josephernest, 2022-08-08:
@mojca 

> What I'm puzzled about: 
> * The source code says "`see hack in /boot/cmline.txt: 38400 is 31250 baud for MIDI!`", but there's no `/boot/cmline.txt` anywhere and > the baud rate was in fact 38400, not 31250 as claimed. Since we control the midi source, we just switched the baud rate to 115200 on > both sides. When sending MIDI messages at 31250 we had tons of strange issues, with midi messaged being misinterpreted at times (while working at other times).

It's a typo, it should be `/boot/cmdline.txt`. See https://github.com/josephernest/SamplerBox/blob/453bbee1f4b3f7d0dc2533e4a3f498bced57d52f/isoimage/samplerbox_iso_maker.sh#L76

See https://www.samplerbox.org/article/midiinwithrpi, point #3.

TL;DR: in older versions of the kernel, for a reason I don't exactly remember, it was impossible to open serial in 31250 baud (as expected for MIDI). Various hacks linked to `bcm2708.uart_clock=3000000`, `init_uart_clock=2441406`, `init_uart_baud=38400` were needed to finally arrive to 31250 baud :)
Now with Linux kernel >= 4.5, this is not needed anymore.

mojca, 2022-08-08:
I cannot check whether it works "with any hardware midi device" (I only have an Arduino), but I'm thinking of doing the reverse test first: try to open a serial connection and send some messages from RPi, while measuring the rate with oscilloscope. Once that gets confirmed, test in the other direction.

That second test sounds a bit more tricky to me because the messages might get interpreted correctly even at a slightly wrong baud rate settings.

By reading #24 it looks like one might have to verify this on:
- various Debian versions
- various RPi models (at least zero, 2B, 4)
- maybe both 32-bit and 64-bit(?)
- using both serial and USB connection
and I'm not sure if I can cover all combinations of them, but I can try some. I won't have hardware access before the weekend (and no hardware + oscilloscope combination before next week).

When things broke for me, it might be that it was a result of some mixture of `apt-get update && apt-get upgrade`, switching between RPi 2B+ and RPi4, ...

What's the oldest OS that you want to target?

On https://www.raspberrypi.com/software/operating-systems/ there is a link to Debian 11 `2022-04-04-raspios-bullseye-armhf-lite.img.xz`, but your scripts are currently pointing to Debian 10.

josephernest, 2022-08-09:
With oscilloscope it might work, and then finally not work on a real Roland, Yamaha synth, etc. So I'll do this test one of these days with real synthesizers, to be sure everything works without the 38400 baud hack :)

rolfmeurer, 2022-08-23:
Python script MIDI IN via SERIAL PORT is very poor. Running status is not considered, so most keyboards produce stuck notes. The Python script then accepts neither NOTE OFF as a NOTE ON message with Velocity=0 nor NOTE OFF as a 0x8n message