josephernest/SamplerBox #50: Last Python2 release

josephernest, 2022-08-08:
@mojca I suggest we make a last Python2 ISO release to make sure everything works for people still using this good old version, and then we can move 100% to Python3 version. Todo:

* check the test at the end of https://github.com/josephernest/SamplerBox/issues/49#issue-1332352471
* if it is confirmed working without this 38400 hack, we remove these things from https://github.com/josephernest/SamplerBox/blob/master/isoimage/root/boot/cmdline.txt and https://github.com/josephernest/SamplerBox/blob/master/isoimage/root/boot/config.txt and the main .py file
* we might simplify https://github.com/josephernest/SamplerBox/blob/master/README.md?plain=1#L22 and https://github.com/josephernest/SamplerBox/blob/master/isoimage/maker.sh#L20, for example: isn't `python-pip` now included out-of-the-box on a RaspiOS Lite like `2021-05-07-raspios-buster-armhf-lite.img`?
* once all these things are checked, I would be interested to see if someone else than me achieves to build an ISO with this: https://github.com/josephernest/SamplerBox/blob/master/isoimage/maker.sh
* if everything is working, we freeze this version into a new release https://github.com/josephernest/SamplerBox/releases

Then we can add features in Python3 branch and make this branch the default one.

mojca, 2022-08-08:
Creating another release before further changes sounds perfectly ok.

But just to clarify: my patches should remain compatible with python 2.7, it's just a matter of picking the python version that gets used by default with your image. The version of python may, but it's not required to be changed immediately.

josephernest, 2022-08-09:
I'm finally rewriting certain parts to include Python3 support directly. I hope to push these next days.

closed 2022-08-09 by josephernest