josephernest/SamplerBox #67: Compatibility with Raspberry Pi 5?

iddryg, 2024-05-15:
I read on a forum post that it's not compatible with Pi 5 yet. Any plan to add that?

josephernest, 2024-05-15:
"Hello"
Any plan to add that, "Thank you in advance"
etc.

Are all these little words no longer used in 2024?  

An open-source creator who puts this project online since years doesn't deserve these simple words of humanity? 

It seems many of us talk to open-source creators and maintainers like we talk to LLM bots. Same here: 
https://github.com/superquadratic/rtmidi-python/issues/25#issuecomment-2077617757
"This project is dead". This is not nice.

Sorry to react this way, but this is the result of many many many emails or issues, where people only say "you should fix this", "you should add this", and no consideration for the people behind the project.

-

Now, the real answer: SamplerBox **is** compatible with Pi 5.
Just start with a RaspiOS, and follow the steps [here](https://github.com/josephernest/SamplerBox?tab=readme-ov-file#install) from step #1, and it will work.

The ready-to-use ISO image, created a few months ago, is probably not working on Pi5, but the project itself **is** compatible with Pi5 if you follow the installation steps.

Have a nice day!

enbeec, 2024-05-15:
> Sorry to react this way, but this is the result of many many many emails or issues, where people only say "you should fix this", "you should add this", and no consideration for the people behind the project.

As someone who has been watching this project for YEARS I would like to step in and validate this.

Manners are good business but also make for better software. Candor and politeness are not a luxury or a bauble! They are ingredients to a successful project and skills that must be maintained. 

Maybe SamplerBox is not a critical piece of software like xzutils but as participants in open source (including people who only contribute to issues by asking questions) we should keep the bar of respect as high as possible. Bullying maintainers is now an attack vector and only the communities can mitigate that.

Thank you for speaking up on your own behalf! And thanks for the project 💕

iddryg, 2024-05-15:
Hi Joseph, 

Sincere apologies, didn't mean to come across short. Admittedly I have been bombarding chatgpt with questions daily. I very much appreciate your years of support on this project and your quick response. I only just came across the project and was interested in purchasing a Pi 5, glad to know it should work! Excited to give it a try. Thank you!